Apollon77/smartmeter-obis #1: Disable transportSerialMessageTimeout

coolchip, 2017-07-31:
Hey. Thanks for your great work.
I use it for a node-red node. My problem is, that your module throws an error, when serial timeout triggers. That stops my complete node-red process. Is it possible to disable the transportSerialMessageTimeout? Set it to null has no effect.

Apollon77, 2017-07-31:
Do you can not catch the exception? I had such timeout cases many times and it only was fixed by starting all over and reopen the serial connection and such.
So maybe make "throw exception on timeout" a new setting where you can choose if you want to have an exception or just ending the read process?!

Apollon77, 2017-08-01:
I have an idea, not to disable to timeout (because I think it's needed), but to get around the "throw an error" topic. Instead I will implement that it simply schedukes the next planned cycle.
Will be hard to test beforehand. Give me some days to find time to do this correct

coolchip, 2017-08-01:
Because you throw this error in the callback of the setTimeout() method, only the timer of nodejs is able to catch it.
The cleanest solution would be to use [Node.js error-first callbacks](http://fredkschott.com/post/2014/03/understanding-error-first-callbacks-in-node-js/). But this would lead to an api change.

coolchip, 2017-08-01:
The usage of your module would then look like this
```javascript
function displayData(err, obisResult) {
    if (err) {
        //handle the error in my own way
        console.log(err);
        return;
    }
    //do something with the data
    console.log(obisResult);
}

var smTransport = SmartmeterObis.init(options, displayData);
```

Apollon77, 2017-08-01:
You are right, this is a good idea ... will look into it for all error cases. Sounds like a 1.0.0 :-)

For now, see 0.6.0 just available now ... 
### v0.6.0 (01.08.2017)
* A serial timeout will no longer trigger an Exception. Instead connection is reset and next cycle is scheduled as configured

coolchip, 2017-08-01:
Thank you very much for that fast fix! :+1:

Apollon77, 2017-08-02:
I will leave it open till 1.0.0 :-)) already working on it

Apollon77, 2017-08-02:
Hey,

if you want you can give 1.0.0-beta1 (also npm) a shot.
Updated README: https://github.com/Apollon77/smartmeter-obis/tree/1.0

I used thae chance to also upgrade to serialport 5.0 which may cause problems and I did not tested with a real serial device so far.
What you think?

coolchip, 2017-08-02:
Hey... this was fast. 
I've tested your beta with my SML reader. Looks good...

```
129-129:199.130.3*255: Manufacturer ID = EMH
1-0:0.0.0*255: Meter owner number = xxxxxxxx
1-0:0.0.9*255: Device ID = xxxxxxxxxxxxxxxxxxxxx
1-0:1.8.0*255: Time integral 1 Sum active power + (Total) = 17223.661 kWh
1-0:2.8.0*255: Time integral 1 Sum active power - (Total) = 0.1649 kWh
1-0:1.8.1*255: Time integral 1 Sum active power + (T1) = 0 kWh
1-0:2.8.1*255: Time integral 1 Sum active power - (T1) = 0.1649 kWh
1-0:1.8.2*255: Time integral 1 Sum active power + (T2) = 17223.661 kWh
1-0:2.8.2*255: Time integral 1 Sum active power - (T2) = 0 kWh
1-0:16.7.0*255: Instantaneous value Total active power (Total) = 419.6 W
129-129:199.130.5*255: Forename/Public Key = xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
```
Don't know the guys from serialport. But it looks like a big step. I think I would wait for 5.0.1 or 5.0.2. :wink:

A little comment to your new functionality with the return in the callback "return true if you want to have the next cycle scheduled...". Thats not the classical node.js way. Normally nobody is interested in the return of a callback. So it feels a little bit strange to return a value that influences your lib. For example: Some (including me) write `return console.log(err)`. Just because it's shorter and nobody cares. :wink:
In my opinion I would cycle the next step, regardless of the return value. If someone wants to stop the cycling he can call `smTransport.stop()`.
But that's just my opinion.

coolchip, 2017-08-02:
If you want to do it anyway, you can pass a callback to your callback:
```javascript
function displayData(err, obisResult, next) {
    if (err) {
        // handle error
        // if you want to have the next cycle scheduled as configured, call next()
        // otherwise process stops
        next();
        return;
    }
    console.log(obisResult);
}
```
But this feels a litte bit overdone.

Apollon77, 2017-08-02:
Thanks for the infos ... really appreciated ... I like the idea that stop is called a bit more ... beta2 on the way soon :-)

Apollon77, 2017-08-02:
For serialport ... they were really good in teh past ... and when they are fast with fixes ... :-)

Apollon77, 2017-08-02:
1.0.0-beta 2 reworked is there ... have fun.

I will install it tomorrow with my project and have it live

Apollon77, 2017-08-02:
I gave it a very fast try and don't worked ... no serial response ... hm ... need to check more in depth

Apollon77, 2017-08-04:
what nodejs version you work with ? 8 or 6?
I think of reverting serialport back to 4.0.7 because seems to compile well unter node 8 too ... hopefully works too :-)

coolchip, 2017-08-04:
I use 8.2.1, because 8 becomes the new long term.

Apollon77, 2017-08-04:
would you please test beta3 (available on npm) ... reverted back to serialport 4.0.7. Thank you!

coolchip, 2017-08-04:
I will try this weekend.

coolchip, 2017-08-07:
Hey, tested your beta3 yesterday.
```
129-129:199.130.3*255: Manufacturer ID = EMH
1-0:0.0.0*255: Meter owner number = xxxxxxxx
1-0:0.0.9*255: Device ID = xxxxxxxxxxxxxxxxxxxx
1-0:1.8.0*255: Time integral 1 Sum active power + (Total) = 17267.8432 kWh
1-0:2.8.0*255: Time integral 1 Sum active power - (Total) = 0.1649 kWh
1-0:1.8.1*255: Time integral 1 Sum active power + (T1) = 0 kWh
1-0:2.8.1*255: Time integral 1 Sum active power - (T1) = 0.1649 kWh
1-0:1.8.2*255: Time integral 1 Sum active power + (T2) = 17267.8432 kWh
1-0:2.8.2*255: Time integral 1 Sum active power - (T2) = 0 kWh
1-0:16.7.0*255: Instantaneous value Total active power (Total) = 204.8 W
129-129:199.130.5*255: Forename/Public Key = xxxxxxxxxxxxxxxxxxxxxxxxxxxxx
```
Looks very fine. 👍 
Also if there is no communication it trys again and again and again... as expected.
```
Error: No or too long answer from Serial Device after last request.
    at SerialResponseTransport.handleSerialTimeout (/home/sebastian/smartmeter-obis-test/node_modules/smartmeter-obis/lib/transports/SerialResponseTransport.js:218:36)
    at Timeout._onTimeout (/home/sebastian/smartmeter-obis-test/node_modules/smartmeter-obis/lib/transports/SerialResponseTransport.js:191:22)
    at ontimeout (timers.js:469:11)
    at tryOnTimeout (timers.js:304:5)
    at Timer.listOnTimeout (timers.js:264:5)
Error: No or too long answer from Serial Device after last request.
    at SerialResponseTransport.handleSerialTimeout (/home/sebastian/smartmeter-obis-test/node_modules/smartmeter-obis/lib/transports/SerialResponseTransport.js:218:36)
    at Timeout._onTimeout (/home/sebastian/smartmeter-obis-test/node_modules/smartmeter-obis/lib/transports/SerialResponseTransport.js:191:22)
    at ontimeout (timers.js:469:11)
    at tryOnTimeout (timers.js:304:5)
    at Timer.listOnTimeout (timers.js:264:5)
Error: No or too long answer from Serial Device after last request.
    at SerialResponseTransport.handleSerialTimeout (/home/sebastian/smartmeter-obis-test/node_modules/smartmeter-obis/lib/transports/SerialResponseTransport.js:218:36)
    at Timeout._onTimeout (/home/sebastian/smartmeter-obis-test/node_modules/smartmeter-obis/lib/transports/SerialResponseTransport.js:191:22)
    at ontimeout (timers.js:469:11)
    at tryOnTimeout (timers.js:304:5)
    at Timer.listOnTimeout (timers.js:264:5)
```
Really nice.

Apollon77, 2017-08-07:
I have it also in production since the weekend.
Because I'm on vacation the next 2 weeks I will wait till I'm back till I officially publish the 1.0.0 ... (just to make sure nothing unexpected happens when have a wider audience using it).
Will leave this issue open till then.
Final 1.0.0 will be same as beta3

coolchip, 2017-08-12:
have a beautiful holiday

coolchip, 2017-08-12:
I also use it and it works fine so far... 😉

Apollon77, 2017-08-25:
So, found a timing problem on some meters and fixed this too... so 1.0.0-beta4 is here ... as last step before 1.0.0 final. If you find time to use it for your project would be cool

coolchip, 2017-08-25:
I still use it at my Project. But only at my home automation. Should I also publish a beta to npm?

Apollon77, 2017-08-25:
SO would be great if you could test in your home with beta4 :-)

if you publish your work as own package is up to you :-))

Apollon77, 2017-09-24:
Prgress :-)
1.0.0 (final) is published and uses official OpenSML 0.3.3 now, so github is gone too

closed 2017-09-24 by Apollon77