Apollon77/smartmeter-obis #4: OMS

martinschki, 2017-12-16:
Hello!
Would it be possible to also read meters with OMS protocol? (https://en.wikipedia.org/wiki/Open_metering_system)
Specifically OMS 3.0.1 Standard Mode 5 with AES encryption

that would be perfect!!!!

Apollon77, 2017-12-16:
Hi,
interesting request. When I understood it correct then 3 topics are relevant:
1.) transport way: MBus wired, wireless or whatever
2.) encryption (AES "mode 5" or whatever with public/private key handling
3.) Data format (DLMS and SML are supported too ... but ...)

This is very hard to implement without having such a device.

Do you have such a device and could provide data (if, I would start with topics 2+3 first)?
I would need a complete set of data (ideally multiple) and at least the used public keys or such, then I could try to get the encryption done somehow and see what's data are inside.

martinschki, 2017-12-18:
I have and I would

I’m just short of the AES key right now
I have requested it from the power company

Apollon77, 2017-12-18:
Thank you! ... I can see if I find time over the christmas time :-) (when you get teh key before they close down for christmas :-) )

deothermo, 2018-01-06:
Integration of OMS would indeed be very interesting.

Please find attached several sample files returned by heat meters as well as by impuls counters to measure gas consumption:
[1.log](https://github.com/Apollon77/smartmeter-obis/files/1608877/1.log)
[2.log](https://github.com/Apollon77/smartmeter-obis/files/1608878/2.log)
[3.log](https://github.com/Apollon77/smartmeter-obis/files/1608879/3.log)
[4.log](https://github.com/Apollon77/smartmeter-obis/files/1608880/4.log)
[5.log](https://github.com/Apollon77/smartmeter-obis/files/1608881/5.log)

Apollon77, 2018-01-06:
Interesting, I awaited OBIS compatible data IDs or such. So maybe it woul be an own lib or I need to rename mine :-) I really wonder that this as "internalional industry standard" comes without a unique ID schema ... hm ... Or is this already parsed somehow to this XML representation?

If it is really only parsing this XML then this is easy ... but what is needed to get this xml representation? :-)

deothermo, 2018-01-07:
Good question. The data was obtained using [libmbus](http://www.rscada.se/libmbus/). There is another java library [here](https://www.openmuc.org/m-bus/).

[This](http://oms-group.org/download4all/oms-spezifikation/) looks like the official documentation for OMS / EN 13757 / mbus / wmbus...

Apollon77, 2018-02-11:
I digged a bit into the topic in the last days.
Communication for OMS is normally be mbus or even wireless-mbus (only?) So this would be a todo.
Additionally all reported data are defined manufactorer specific and without any OBIS IDs or such, just "several data" ... the XML above shows really good what as data are available.

libmbus is currently not caring about the data decryption.

Apollon77, 2018-03-06:
Hey all,

I made some progress.
I was able to read un-encrypted mbus and created a very new nodejs library for this that could be used as base. So if you would like to support I would be happy if you can test this - also for encrypted payload this test would be great to see what happends and how the underlying "libmbus" reacts on encrypted payload.

create a new directory on your computer and run

npm install node-mbus --debug --save

then you can find in node_modules/node-mbus/example.js a small example script to query data when you know the mbus id. Do you know it?

if not you would need to change the call from "getData" to "scanSecondary" (and remove the first parameter, so only have the callback) to scan the mbus network for the secondary ID. you can use this then for getData.

The "--debug" on npm install above will enable some additional serial logging to exactly see the messages and such. So it would be great (independently from the result) if you can send the output :-)

This could be the base for a way to read it in general ... first step to include in the library

Apollon77, 2018-05-18:
Also added more links and infos:
https://github.com/volkszaehler/vzlogger/pull/181/files
https://www.symcon.de/forum/threads/30174-Smartmeter-auslesen-Verschlüsselung/page2?s=91c4f638bb31c59d72dd2a72e2c46ce5
https://github.com/volkszaehler/vzlogger/issues/176

mschlenstedt, 2019-01-03:
Hi,

I implemented SML and D0 rudimentary myself to  small piece of software for providing meter data to a smarthome system: https://www.loxforum.com/forum/projektforen/loxberry/plugins/85702-neues-plugin-smartmeter-stromz%C3%A4hler-auslesen

I plan to change over to your fantastic library, because it's much better than what I did :-) Now my question: I have also a lot of requests for (I guess) OMS support with AES encryption, e. g. Siemens TD-3511 https://wiki.volkszaehler.org/hardware/channels/meters/power/edl-ehz/siemens_td3511_in_oberoesterreich

A lot of people would provide SSH-Access to a Linux System connected to such a device. Would that help to implement this protocol to your lib?

Apollon77, 2019-01-10:
Basically interesting yes, but the topic is really compex and time is rare :-( I do not know when I will find time to look into this topic :-(

Apollon77, 2019-12-23:
also: https://www.netzooe.at/NOOE-AMIS-Bedienungsanleitung-03-2019.pdf?ch=1DitW6AD&:hp=3;2;de

// https://www.netzooe.at/Themen/Information/smart-meter/FAQ

* Physikalische Schicht: Infrarotschnittstelle (IEC 62056-21)
* Sicherungs-, Vermittlungs- und Anwendungsschicht: M-Bus (EN 13757-1 bis EN 13757-3).
* Verschlüsselung nach OMS 3.0.1: AES128 Encryption Mode 5