Apollon77/smartmeter-obis #5: StdInOutTransport option ?

devZer0, 2018-01-02:
According to standard Unix Philosophy *), despite a file name normally one can use "-" as filename for a unix tool and then stdin/out is being used. 

Unfortunately, it doesn`t work with smartmeter-obis/FileTransport

Could you provide sort of a "StdInOutTransport" option ?

*) "https://en.wikipedia.org/wiki/Unix_philosophy
 This is the Unix philosophy: Write programs that do one thing and do it well. Write programs to work together. Write programs to handle text streams, because that is a universal interface.""

Apollon77, 2018-01-02:
But StdIn (so "read only") would be sufficient or ?!

devZer0, 2018-01-02:
yes i think so

Apollon77, 2018-01-02:
Then see 1.1.0-alpha 5 available on npm now :-) Doc is current too. Use StdInTransport. tests with a mocked stdin are successfull ...
Please provide Feedback

Apollon77, 2018-01-02:
Update, try 1.1.0-alpha6, should now also support "stopping" on EOF on stdin

devZer0, 2018-01-21:
works fantastic! thanks!

 # find /root/libsml-testing/*.bin|while read line;do echo $line; cat $line | nodejs StdInSml.js|head -n 30  ;echo ;done 
/root/libsml-testing/DrNeuhaus_SMARTY_ix-130.bin
129-129:199.130.3*255: Manufacturer ID = DNT
1-0:0.0.9*255: Device ID = 0901444e540100002030
1-0:2.8.0*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (Total) = 4.373 kWh
1-0:2.8.1*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T1) = 4.373 kWh
1-0:2.8.2*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T2) = 0 kWh
1-0:15.7.0*255: Momentanwert (Total) = 0 W
129-129:199.130.5*255: Forename/Public Key = fea5d6bc658d734673f5790051d61c2ebd55344a6285686de3b645ee0104d65f5abba6cb68f1bfcb7fcf94f99cd4d845
129-129:199.130.3*255: Manufacturer ID = DNT
1-0:0.0.9*255: Device ID = 0901444e540100002030
1-0:2.8.0*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (Total) = 4.373 kWh
1-0:2.8.1*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T1) = 4.373 kWh
1-0:2.8.2*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T2) = 0 kWh
1-0:15.7.0*255: Momentanwert (Total) = 0 W
129-129:199.130.5*255: Forename/Public Key = fea5d6bc658d734673f5790051d61c2ebd55344a6285686de3b645ee0104d65f5abba6cb68f1bfcb7fcf94f99cd4d845
129-129:199.130.3*255: Manufacturer ID = DNT
1-0:0.0.9*255: Device ID = 0901444e540100002030
1-0:2.8.0*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (Total) = 4.373 kWh
1-0:2.8.1*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T1) = 4.373 kWh
1-0:2.8.2*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T2) = 0 kWh
1-0:15.7.0*255: Momentanwert (Total) = 0 W
129-129:199.130.5*255: Forename/Public Key = fea5d6bc658d734673f5790051d61c2ebd55344a6285686de3b645ee0104d65f5abba6cb68f1bfcb7fcf94f99cd4d845
129-129:199.130.3*255: Manufacturer ID = DNT
1-0:0.0.9*255: Device ID = 0901444e540100002030
1-0:2.8.0*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (Total) = 4.373 kWh
1-0:2.8.1*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T1) = 4.373 kWh
1-0:2.8.2*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T2) = 0 kWh
1-0:15.7.0*255: Momentanwert (Total) = 0 W
129-129:199.130.5*255: Forename/Public Key = fea5d6bc658d734673f5790051d61c2ebd55344a6285686de3b645ee0104d65f5abba6cb68f1bfcb7fcf94f99cd4d845
129-129:199.130.3*255: Manufacturer ID = DNT
1-0:0.0.9*255: Device ID = 0901444e540100002030

/root/libsml-testing/EMH_eHZ-GW8E2A500AK2.bin
129-129:199.130.3*255: Manufacturer ID = EMH
1-0:0.0.0*255: Meter owner number = 02280816
1-0:1.8.1*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T1) = 14798.1129 kWh
1-0:1.8.2*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T2) = 2.0124 kWh
0-0:96.1.255*255: Factory number = 0002280816
1-0:1.7.0*255: Momentanwert Summe Wirkarbeit Bezug + (Total) = 13.8 W
129-129:199.130.3*255: Manufacturer ID = EMH
1-0:0.0.0*255: Meter owner number = 02280816
1-0:1.8.1*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T1) = 14798.113 kWh
1-0:1.8.2*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T2) = 2.0124 kWh
0-0:96.1.255*255: Factory number = 0002280816
1-0:1.7.0*255: Momentanwert Summe Wirkarbeit Bezug + (Total) = 14.7 W
129-129:199.130.3*255: Manufacturer ID = EMH
1-0:0.0.0*255: Meter owner number = 02280816
1-0:1.8.1*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T1) = 14798.113 kWh
1-0:1.8.2*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T2) = 2.0124 kWh
0-0:96.1.255*255: Factory number = 0002280816
1-0:1.7.0*255: Momentanwert Summe Wirkarbeit Bezug + (Total) = 14.9 W
129-129:199.130.3*255: Manufacturer ID = EMH
1-0:0.0.0*255: Meter owner number = 02280816
1-0:1.8.1*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T1) = 14798.113 kWh
1-0:1.8.2*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T2) = 2.0124 kWh
0-0:96.1.255*255: Factory number = 0002280816
1-0:1.7.0*255: Momentanwert Summe Wirkarbeit Bezug + (Total) = 14.2 W
129-129:199.130.3*255: Manufacturer ID = EMH
1-0:0.0.0*255: Meter owner number = 02280816
1-0:1.8.1*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T1) = 14798.113 kWh
1-0:1.8.2*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T2) = 2.0124 kWh
0-0:96.1.255*255: Factory number = 0002280816
1-0:1.7.0*255: Momentanwert Summe Wirkarbeit Bezug + (Total) = 14.8 W

/root/libsml-testing/EMH_eHZ-HW8E2A5L0EK2P_1.bin
129-129:199.130.3*255: Manufacturer ID = EMH
1-0:0.0.9*255: Device ID = 06454d4801001d4615ca
[... 99 lines cut ...]
1-0:16.7.0*255: Momentanwert Gesamtwirkleistung (Total) = 136.7 W
129-129:199.130.5*255: Forename/Public Key = 8b6a0e6e12f5d980f730b6bd5e1941834eb0e43e4a6323d999259556f5e56e040498c89738f0f6dff8785b045d84e0d6
1-0:96.50.2*4: Wartungseintrag (T2) = 637
1-0:96.50.2*6: Wartungseintrag (T2) = undefined
129-129:199.130.3*255: Manufacturer ID = EMH
1-0:0.0.9*255: Device ID = 06454d480107197c2456
1-0:1.8.0*255: Zählerstand 1 Summe Wirkarbeit Bezug + (Total) = 2795.6928 kWh
1-0:1.8.1*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T1) = 2795.6928 kWh
1-0:1.8.2*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T2) = 0 kWh
1-0:16.7.0*255: Momentanwert Gesamtwirkleistung (Total) = 136.8 W
129-129:199.130.5*255: Forename/Public Key = 8b6a0e6e12f5d980f730b6bd5e1941834eb0e43e4a6323d999259556f5e56e040498c89738f0f6dff8785b045d84e0d6
1-0:96.50.2*4: Wartungseintrag (T2) = 637
1-0:96.50.2*6: Wartungseintrag (T2) = undefined
129-129:199.130.3*255: Manufacturer ID = EMH
1-0:0.0.9*255: Device ID = 06454d480107197c2456
1-0:1.8.0*255: Zählerstand 1 Summe Wirkarbeit Bezug + (Total) = 2795.6929 kWh
1-0:1.8.1*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T1) = 2795.6929 kWh
1-0:1.8.2*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T2) = 0 kWh
1-0:16.7.0*255: Momentanwert Gesamtwirkleistung (Total) = 136.5 W
129-129:199.130.5*255: Forename/Public Key = 8b6a0e6e12f5d980f730b6bd5e1941834eb0e43e4a6323d999259556f5e56e040498c89738f0f6dff8785b045d84e0d6
1-0:96.50.2*4: Wartungseintrag (T2) = 637
1-0:96.50.2*6: Wartungseintrag (T2) = undefined
129-129:199.130.3*255: Manufacturer ID = EMH
1-0:0.0.9*255: Device ID = 06454d480107197c2456
1-0:1.8.0*255: Zählerstand 1 Summe Wirkarbeit Bezug + (Total) = 2795.693 kWh

closed 2018-01-21 by devZer0

Apollon77, 2018-01-21:
have published alpha12 earlier today, contains a bugfix ... when it now runs without new problems I will publish 1.1.0 oficially latest next weekend

devZer0, 2018-01-21:
when feeding sml via "cat /dev/ttyUSB0 | nodejs stdin.js" and did test that a little bit, but getting 

ERROR: Error: Message CRC-Check invalid -> no data available

from time to time and everything stops then.

still trying to reproduce/finding a pattern

devZer0, 2018-01-21:
i did 
cat /dev/ttyUSB0 | tee /tmp/ttyUSB0.dump | nodejs stdin.js 
and got CRC error after some time.

when feeding ttyUSB.dump via stdin into smartmeter-obis , it get's reproducably stuck after 278kb

cat /tmp/ttyUSB0.dump |pv -bc |nodejs ./StdInSml.js >/dev/null 
 278kB
^C

can you reproduce that?

[ttyUSB0.dump.gz](https://github.com/Apollon77/smartmeter-obis/files/1650149/ttyUSB0.dump.gz)

devZer0, 2018-01-21:
oops, apparently 278kB is the size of the dump and CRC error is not shown with that, so the question is why i did get that error with cat /dev/ttyUBS0 | tee /tmp/ttyUSB0.dump | ... but not with cat /tmp/ttyUSB0.dump   - and why it hangs when instead of terminating (as there is EOF detection)

Apollon77, 2018-01-21:
What value you have set for the transportStdInMessageTimeout ? After this time he should end in any case. it would be interesting to see the "debug:2" output for this. I will need some time to try to reproduce for ym own ... else you send the end of the debug output :-)

I fixed one issue in StdInTransport ... try alpha13 please and ideally send the last 20 lines from debug output.

CRC error happens when the crc check of the SML reports that error :-) When CRC error happens normally only this one message is discarded. There is a flag to ignore CRC errors ...

devZer0, 2018-01-21:
i added transportStdInMessageTimeout: 2000 - but it does not end:

1-0:15.7.0*255: Momentanwert (Total) = 44 W
129-129:199.130.5*255: Forename/Public Key = fea5d6bc658d734673f5790051d61c2ebd55344a6285686de3b645ee0104d65f5abba6cb68f1bfcb7fcf94f99cd4d845
129-129:199.130.3*255: Manufacturer ID = DNT
1-0:0.0.9*255: Device ID = 0901444e540100002030
1-0:2.8.0*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (Total) = 4.373 kWh
1-0:2.8.1*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T1) = 4.373 kWh
1-0:2.8.2*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T2) = 0 kWh
1-0:15.7.0*255: Momentanwert (Total) = 41 W
129-129:199.130.5*255: Forename/Public Key = fea5d6bc658d734673f5790051d61c2ebd55344a6285686de3b645ee0104d65f5abba6cb68f1bfcb7fcf94f99cd4d845
ERROR: Error: No or too long answer from StdIn after last request.
ERROR: Error: No or too long answer from StdIn after last request.
ERROR: Error: No or too long answer from StdIn after last request.

with debug=2 runtime was very very long:

	Obj-Name: 129-129:199.130.5*255
	Status: undefined
	Val-Time: 
	Unit: undefined
	Scaler: undefined
	Value: ��ּe�sFs�yQ�.�U4Jb�hm�E��_Z���h��ϔ����E / fea5d6bc658d734673f5790051d61c2ebd55344a6285686de3b645ee0104d65f5abba6cb68f1bfcb7fcf94f99cd4d845
	Value-Signature: undefined
, 
]
			List-Signature: undefined
			Act-Gateway-Time: 
		CRC 16: valid
	END SmlMessage

	START SmlMessage
		Transaction-ID:  (00001b56)
		Group-No: 0
		Abort On Error: 0
		Message-Body: 		SmlPublicCloseResponse
			Global Signature: 
		CRC 16: valid
	END SmlMessage

END SML-File

STORE DATA
129-129:199.130.3*255: Manufacturer ID = DNT
1-0:0.0.9*255: Device ID = 0901444e540100002030
1-0:2.8.0*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (Total) = 4.373 kWh
1-0:2.8.1*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T1) = 4.373 kWh
1-0:2.8.2*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T2) = 0 kWh
1-0:15.7.0*255: Momentanwert (Total) = 41 W
129-129:199.130.5*255: Forename/Public Key = fea5d6bc658d734673f5790051d61c2ebd55344a6285686de3b645ee0104d65f5abba6cb68f1bfcb7fcf94f99cd4d845
LEFT AFTER HANDLE-MESSAGE 28
SET MESSAGE TIMEOUT TIMER: 2000
RESUME READING STDIN
REMAINING DATA AFTER MESSAGE HANDLING: vbbrcv	
MORE DATA AVAILABLE, PROCESS THEM 28
MATCH-RESULT MESSAGE: "1b1b1b1b01010101760500001b576200620072630101760101051d09" -> null
STDIN RESUME
SET MESSAGE TIMEOUT TIMER: 2000
MESSAGE TIMEOUT TRIGGERED
Error: No or too long answer from StdIn after last request.
    at StdInTransport.handleStdInTimeout (/root/node_modules/smartmeter-obis/lib/transports/StdInTransport.js:229:36)
    at Timeout._onTimeout (/root/node_modules/smartmeter-obis/lib/transports/StdInTransport.js:200:14)
    at ontimeout (timers.js:475:11)
    at tryOnTimeout (timers.js:310:5)
    at Timer.listOnTimeout (timers.js:270:5)
ERROR: Error: No or too long answer from StdIn after last request.
Transport Reset!! Restart = true
SCHEDULE NEXT RUN IN 0s
STDIN RESUME
SET MESSAGE TIMEOUT TIMER: 2000
^C

Apollon77, 2018-01-21:
that it runs longer with debug:2 is normal because of the mass of outputs. Do you have captured the output somehow? question is if you have the line "STDIN END" somewhere? He tries to read further, so if the EOF was not received ... or your store functions resets it. Can you post your store method code please.

devZer0, 2018-01-21:
i attached the binary dump above, so you should be able to feed it in yourself.

i don't see any "STDIN END" in the output.

below is my program. 

oh, btw - you have a lot of test programs added, what about adding example programs for every sort of use case? that would make it easier for beginners to start with smartmeter-obis...

StdInSml.js 
        var SmartmeterObis = require('../index.js');

        var options = {
            'protocol': 'SmlProtocol',
            'transport': 'StdInTransport',
            'requestInterval': 0,
            'obisNameLanguage': 'de',
            'transportStdInMaxBufferSize': 1000000,
            'debug': 2 
        };

        var lastObisResult;
        var counter = 0;
        var errCounter = 0;

        function displayData(err, obisResult) {
            if (err) {
                console.log('ERROR: ' + err);
                return;
            }
            for (var obisId in obisResult) {
                console.log(obisResult[obisId].idToString() + ': ' + SmartmeterObis.ObisNames.resolveObisName(obisResult[obisId], options.obisNameLanguage).obisName + ' = ' + obisResult[obisId].valueToString());
            }
        }

        var smTransport = SmartmeterObis.init(options, displayData);

        smTransport.process();

Apollon77, 2018-01-22:
Can it be that with „piping in the file“ no eof is sent?! Seems so to me. What is when you use < Filename to feed it? Or use the other transport for files :-)) else the lib has no way to detect feof. 
Alternatively you can check the timeout error in the callback method to stop as needed. 

If you have a stdin example we could create an example folder with the example script. I‘m happy to get a PR ;-)

devZer0, 2018-01-22:
it must have something to do with the length of the file or contents inside, because with all my short binary dumps from https://github.com/devZer0/libsml-testing it works without a problem end EOF is detected, even with feeding with cat filename | ... -  it only gets stuck when using the 278kB dump where the CRC error occured. so please have a look at that dump, i have attached it.

furthermore, i found http://pubs.opengroup.org/onlinepubs/9699919799/functions/read.html
-snipp-
When attempting to read from an empty pipe or FIFO:
If no process has the pipe open for writing, read() shall return 0 to indicate end-of-file.
-snipp-

Apollon77, 2018-01-22:
In your case the effect is that there are 28 bytes from a new message in the buffer and so he waits to receive the rest of it. Then he runs into the timeout after 2 secs not receiving the next bytes and you get the "Serial timeout" error. in this case you do not stop it but tell to continue and so he still waits on input data from the stdin.
nodejs provides an "end" event on the stdin stream to show that EOF happened. When this is there in debug:2 you get that "STDIN END" debug message (that's why I asked). But as you told it do not appear - this means for me that I have no chance to detect the EOF (ecause nodejs did not send the event ... whyever).

try one thing please. remove the last 28 bytes from your file to have the end at a "full" message. And try to process then. Do it stops or still waits for more input?

Apollon77, 2018-01-22:
And make sure the internal buffer is big enough to hold 278 kb !! Or check the debug output for BUFFER OVERRUN errors!

Apollon77, 2018-01-25:
I have did some flow optimizations (don't know if they in the end also really affected Stdin) ... but try alpha14 ... Serial is much more stable

devZer0, 2018-01-25:
i tried to compare StdIn vs. Serial Transport and found the following issue: https://github.com/Apollon77/smartmeter-obis/issues/7

Apollon77, 2018-01-25:
See other issue. Really used alpha14 or latest github code?

Apollon77, 2018-01-25:
alpha15 is your friend :-) stupid bug ...
And the virtual-serialport library that was used in the tests also send out data if port was never opened, so tests did not showed that :-( But this is fixed too now

devZer0, 2018-01-25:
regaring the stdin issue with larger data amount - i tried to remove the remaining 28 bytes so the last sml message is complete, but after that debug still showed "LEFT AFTER HANDLE-MESSAGE 28"

if i remove some bytes more - there is still 28 bytes left. 

so i guess there is problematic data in the stream at an earlier point in time.

i wonder why you can't reproduce - did you try at all? binary dump is attached in this thread...

Apollon77, 2018-01-25:
don't had time so far. Please send your script that you use for that piping and I will try it tomorrow

devZer0, 2018-01-26:
its already in this thread

Apollon77, 2018-01-26:
I tested around many things and I have no idea why this is not working but I can reproduce it.

Am "echo | ..." brings

```
ASSIGN STDIN
STDIN RESUME
SET MESSAGE TIMEOUT TIMER: 120000
ADD NEW DATA (0 + NEW 1)
MATCH-RESULT MESSAGE: "0a" -> null
STDIN END
```

a cat of an file with only a line break also brings correct behaviour:
```
cat txtin.txt |node directStdInSml.js
ASSIGN STDIN
STDIN RESUME
SET MESSAGE TIMEOUT TIMER: 120000
ADD NEW DATA (0 + NEW 2)
MATCH-RESULT MESSAGE: "0a0a" -> null
STDIN END
```
... so an "EOF" is send and received correctly in general. I also tested with several filesizes. It seems that at a certain point of size of the file that is piped into the nodejs process do not get an EOF anymore or at least do not report it as event.
But also google did't helped, so I'm finished for now ... 

Yes there is a problem, but I have no way to fix it for now, sorry

Apollon77, 2018-01-27:
I checked one more thing and I think that the main reason is something different (as you also noticed in one comment above).
With me the stdin gets 4x65536 bytes = 262.144 bytes ... but your file is bigger! So nodejs stops to emit "data" events and because the file is not complete also no "end" is there ... Why ... still unclear. No idea ... :-(

Apollon77, 2018-01-28:
So I again invested some time ... Effect stays the same also when I use readable streams and such.

A minimalistic script with only communication will work and get all data including the end event.

I still have no idea what happens there :-(

devZer0, 2018-01-29:
mhh, thanks for investigating. maybe it's a nodejs bug?

Apollon77, 2018-01-29:
To find that out I tried with a minimal script, but there all data are delivered. So it's somehow a combination ... but I have no idea what I can do without completely comment out my code and put back in one piece after the otehr. But sorry for this I currently don't have the time :-(( Will leave it on my list