Apollon77/smartmeter-obis #6: EMH ITZ meter

deothermo, 2018-01-05:
Hi,

thanks for the great work!

I have a EMH ITZ meter which is connected through an USB optical probe (300bd, 7E1). I am currently using the node-red contribution [smartmeter](https://www.npmjs.com/package/node-red-contrib-smartmeter) to acquire the data. 

However, the obis data returned by the meter can currently not be interpreted due to the &-sign after the obis code (e.g. "0.1.2&07(0180105115851)"). A complete dump can be found [here.txt](https://github.com/Apollon77/smartmeter-obis/files/1607690/obis.txt)

The [obis documentation](http://www.edi-energy.de/files2%5COBIS-Kennzahlen-System%202.2a_20130401.pdf) requires a "*", however EMH apparently places a "&" for some data sets.

Have you come across this with other meters?

Apollon77, 2018-01-05:
Interesting. No I think this is new. 

Protocol used is D0? Then the fix should be easy

Apollon77, 2018-01-05:
try 1.1.0-alpha7, available on npm

closed 2018-01-05 by Apollon77

deothermo, 2018-01-05:
Yes, protocol is D0

deothermo, 2018-01-05:
excellent, works!

dhirajthaokar, 2018-04-07:
[image]
i have Secure EHL meter

Apollon77, 2018-04-07:
Please create a new issue ... because you do not have an EMH ITZ meter ... you have something different and still the question is how to read data at all. You need to find that out

commit 63dca243 (2018-01-05): 1.1.0-alpha7 fixes #6