Apollon77/smartmeter-obis #7: SerialResponseTransport with requestInterval=0 does not work anymore

devZer0, 2018-01-25:
getting no output anymore with the program below

i have taken a look and apparently commit 1191031aefe565d4f2a23cb455bd2aac607c03b4 must have broken something

nb-cq57 smartmeter-obis # nodejs Serial.js 
129-129:199.130.3*255: Manufacturer ID = DNT
1-0:0.0.9*255: Device ID = 0901444e540100002030
1-0:2.8.0*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (Total) = 4.373 kWh
1-0:2.8.1*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T1) = 4.373 kWh
1-0:2.8.2*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T2) = 0 kWh
1-0:15.7.0*255: Momentanwert (Total) = 17 W
129-129:199.130.5*255: Forename/Public Key = fea5d6bc658d734673f5790051d61c2ebd55344a6285686de3b645ee0104d65f5abba6cb68f1bfcb7fcf94f99cd4d845
129-129:199.130.3*255: Manufacturer ID = DNT
1-0:0.0.9*255: Device ID = 0901444e540100002030
1-0:2.8.0*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (Total) = 4.373 kWh
1-0:2.8.1*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T1) = 4.373 kWh
1-0:2.8.2*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T2) = 0 kWh
1-0:15.7.0*255: Momentanwert (Total) = 18 W
129-129:199.130.5*255: Forename/Public Key = fea5d6bc658d734673f5790051d61c2ebd55344a6285686de3b645ee0104d65f5abba6cb68f1bfcb7fcf94f99cd4d845
^C
nb-cq57 smartmeter-obis # git checkout 1191031aefe565d4f2a23cb455bd2aac607c03b4
Vorherige Position von HEAD war bbac81d... updates to support requestIntervall = -1 to process only one message
HEAD ist jetzt bei 1191031... use 1.1.0-alpha11 from lib
nb-cq57 smartmeter-obis # vi Serial.js 
nb-cq57 smartmeter-obis # nodejs Serial.js 
CREATE SERIALPORT: 9600 8 1 none
SET MESSAGE TIMEOUT TIMER: 10000
MESSAGE TIMEOUT TRIGGERED
Error: No or too long answer from Serial Device after last request.
    at SerialResponseTransport.handleSerialTimeout (/root/smartmeter-obis/lib/transports/SerialResponseTransport.js:275:36)
    at Timeout._onTimeout (/root/smartmeter-obis/lib/transports/SerialResponseTransport.js:233:18)
    at ontimeout (timers.js:475:11)
    at tryOnTimeout (timers.js:310:5)
    at Timer.listOnTimeout (timers.js:270:5)
error
Transport Reset!! Restart = true
ERROR CLOSING SERIALPORT
SCHEDULE NEXT RUN IN 0s
CREATE SERIALPORT: 9600 8 1 none
SET MESSAGE TIMEOUT TIMER: 10000
^C

var SmartmeterObis = require('../index.js');

var options = {
    'protocol': "SmlProtocol",
    'transport': "SerialResponseTransport",
    'transportSerialPort': "/dev/ttyUSB0",
    'transportSerialBaudrate': 9600,
    'requestInterval': 0, 
    'obisNameLanguage': 'de',
};

function displayData(err, obisResult) {
if (err) {
	  console.log("error");
	  return;
        }  
    for (var obisId in obisResult) {
       console.log(obisResult[obisId].idToString() + ": " + SmartmeterObis.ObisNames.resolveObisName(obisResult[obisId], options.obisNameLanguage).obisName + ' = ' + obisResult[obisId].valueToString());
   }
}

var smTransport = SmartmeterObis.init(options, displayData);

smTransport.process();

Apollon77, 2018-01-25:
You habe tried latest alpha14?

Apollon77, 2018-01-25:
Sorry close was not planned. And yes in between there were commits that broke things. Alpha14 should work. If not I need to retest.

Apollon77, 2018-01-25:
Ok will check later :-(

Apollon77, 2018-01-25:
is the last output from "debug:2" ? If not please use that and send log.
In the tests evereything works as expected

Apollon77, 2018-01-25:
I think I found it ...

closed 2018-01-25 by Apollon77

devZer0, 2018-01-25:
fantastico! seems that it's fixed!
that was quick, thanks!

commit c3225cd0 (2018-01-25): alpha15, fixes #7