Apollon77/smartmeter-obis #15: DZG DWS7412 meter

kersten, 2018-03-05:
Hi,

thanks for the great work!

My smartmeter is working too: DZG DWS7412.1T

Greetings

Apollon77, 2018-03-06:
Thank you, I will note it on the readme

kersten, 2018-03-06:
But I only get one number. Only the sum amount of used energy. I need to further check if I can get more out of it. Maybe you could help me with that. I could post a debug log later.

Apollon77, 2018-03-06:
Post debug and I will check. It really depends on the power company how they configued the smartmeter because they define which data will be send by the smartmeter in the end ... and yes it can be that the display shows more then delivered using D0/SML ... :-(

kersten, 2018-03-06:
Ah ok, I see. What a shame!

Apollon77, 2018-03-06:
BTW: Is it D0 or SML?
If D0 there may be additional commands beside the standart one "/?" that bring more/other data ...

kersten, 2018-03-06:
No, the data comes in SML file format

kersten, 2018-03-06:
Hi,

this is the debug output:

```SERIALPORT OPEN
SET MESSAGE TIMEOUT TIMER: 120000
ADD NEW DATA (0 + NEW 15)
MATCH-RESULT MESSAGE: "5a470177070100600100ff01726201" -> null
ADD NEW DATA (15 + NEW 14)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a470002" -> null
ADD NEW DATA (29 + NEW 14)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c" -> null
ADD NEW DATA (43 + NEW 13)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e52036248" -> null
ADD NEW DATA (56 + NEW 15)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e520362480101016344cf00760560bf12006200" -> null
ADD NEW DATA (71 + NEW 14)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e520362480101016344cf00760560bf1200620062007263020171016351f5000000" -> null
ADD NEW DATA (85 + NEW 10)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e520362480101016344cf00760560bf1200620062007263020171016351f500000000001b1b1b1b1a04acfc" -> null
ADD NEW DATA (95 + NEW 12)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e520362480101016344cf00760560bf1200620062007263020171016351f500000000001b1b1b1b1a04acfc1b1b1b1b01010101760561bf" -> null
ADD NEW DATA (107 + NEW 14)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e520362480101016344cf00760560bf1200620062007263020171016351f500000000001b1b1b1b1a04acfc1b1b1b1b01010101760561bf1200620062007263010176010102" -> null
ADD NEW DATA (121 + NEW 15)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e520362480101016344cf00760560bf1200620062007263020171016351f500000000001b1b1b1b1a04acfc1b1b1b1b01010101760561bf1200620062007263010176010102310b0a01445a4700029f7f72726201" -> null
ADD NEW DATA (136 + NEW 14)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e520362480101016344cf00760560bf1200620062007263020171016351f500000000001b1b1b1b1a04acfc1b1b1b1b01010101760561bf1200620062007263010176010102310b0a01445a4700029f7f72726201640655886202630dc000760562bf" -> null
ADD NEW DATA (150 + NEW 13)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e520362480101016344cf00760560bf1200620062007263020171016351f500000000001b1b1b1b1a04acfc1b1b1b1b01010101760561bf1200620062007263010176010102310b0a01445a4700029f7f72726201640655886202630dc000760562bf1200620062007263070177010b" -> null
ADD NEW DATA (163 + NEW 15)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e520362480101016344cf00760560bf1200620062007263020171016351f500000000001b1b1b1b1a04acfc1b1b1b1b01010101760561bf1200620062007263010176010102310b0a01445a4700029f7f72726201640655886202630dc000760562bf1200620062007263070177010b0a01445a4700029f7f72070100620a" -> null
ADD NEW DATA (178 + NEW 14)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e520362480101016344cf00760560bf1200620062007263020171016351f500000000001b1b1b1b1a04acfc1b1b1b1b01010101760561bf1200620062007263010176010102310b0a01445a4700029f7f72726201640655886202630dc000760562bf1200620062007263070177010b0a01445a4700029f7f72070100620affff726201640655887377070100" -> null
ADD NEW DATA (192 + NEW 13)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e520362480101016344cf00760560bf1200620062007263020171016351f500000000001b1b1b1b1a04acfc1b1b1b1b01010101760561bf1200620062007263010176010102310b0a01445a4700029f7f72726201640655886202630dc000760562bf1200620062007263070177010b0a01445a4700029f7f72070100620affff72620164065588737707010060320101017262016200620052" -> null
ADD NEW DATA (205 + NEW 14)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e520362480101016344cf00760560bf1200620062007263020171016351f500000000001b1b1b1b1a04acfc1b1b1b1b01010101760561bf1200620062007263010176010102310b0a01445a4700029f7f72726201640655886202630dc000760562bf1200620062007263070177010b0a01445a4700029f7f72070100620affff726201640655887377070100603201010172620162006200520004445a470177070100600100ff" -> null
ADD NEW DATA (219 + NEW 15)
MATCH-RESULT MESSAGE: "5a470177070100600100ff017262016200620052000b0a01445a4700029f7f720177070100010800ff641c01047262016200621e520362480101016344cf00760560bf1200620062007263020171016351f500000000001b1b1b1b1a04acfc1b1b1b1b01010101760561bf1200620062007263010176010102310b0a01445a4700029f7f72726201640655886202630dc000760562bf1200620062007263070177010b0a01445a4700029f7f72070100620affff726201640655887377070100603201010172620162006200520004445a470177070100600100ff017262016200620052000b0a01445a" -> null
ADD NEW DATA (234 + NEW 13)
[... cut ...]
	END SmlMessage

	START SmlMessage
		Transaction-ID: c� (63bf1200)
		Group-No: 0
		Abort On Error: 0
		Message-Body: 		SmlPublicCloseResponse
			Global Signature:
		CRC 16: valid
	END SmlMessage

END SML-File

STORE DATA
1-0:96.50.1*1: Wartungseintrag (T1) letztes Jahr = DZG
1-0:96.1.0*255: Serialnumber = 0a01445a4700029f7f72
1-0:1.8.0*255: Zählerstand 1 Summe Wirkarbeit Bezug + (Total) = 72 kWh
SET MESSAGE TIMEOUT TIMER: 120000
RESUME READING SERIALPORT
SCHEDULE NEXT RUN IN 10s
SERIALPORT CLOSE`

Apollon77, 2018-03-07:
The output at the end shows exactly what is returned ... :-(

kersten, 2018-03-07:
sad, so no currently served watts?

Apollon77, 2018-03-07:
nope ... you need to talk to your power company (but most likely they do not change these things for singe customers).

closed 2018-03-14 by Apollon77