Apollon77/smartmeter-obis #21: how to communicate secure apex meter

dhirajthaokar, 2018-04-09:
[image]
[image]

Apollon77, 2018-04-09:
I googled around but for this type (on other photo was a type name) I was not able to find anything on infos. Do you have a datasheet of this device?
is the reading head correctly mounted?

dhirajthaokar, 2018-04-09:
its a  oldest meter  now will we used seucre apex 100.

Apollon77, 2018-04-09:
For that it reads "standard  optical interface" which should match to the D0Protocol, SerialRequestResponseTransport and no special  parameters needed. But when you do not get anything as answer (even no "crap data" which would point to s wrong baudrate or such)then I can not help.
Check the IR-Read/Write haed that he works, is attached/mounted correctly

dhirajthaokar, 2018-04-11:
I cheked on port sensor using smartphone camera when programme sens request.
Optical probe is ok
I am using 300 bps

Apollon77, 2018-04-11:
When you just do a „cat“ on the device do you get data on normal console?

closed 2018-05-18 by Apollon77