Apollon77/smartmeter-obis #25: Elster As1440 with rs485

ublsoftware, 2018-05-02:
congratulations to all for the work done !! I saw that the library is compatible with Elster As1440, also works with RS485 cable or only with infrared? can you possibly help me?
Thank you
Roberto

Apollon77, 2018-05-02:
IR works in any case with a "ir-read/write head". What exactly do you mean with RS485?

The ir-heads send/receive data over a USB "converter" in a serial way. When RS485 works the same and the transport layer seen from a computer is the same then it could work too ... but no experience/feedback from this so far.

So you need to provide more information on this

ublsoftware, 2018-05-03:
thanks, I mean with direct connection via RS485 pin of the AS1440 without infrared
Roberto

Apollon77, 2018-05-03:
Yes I understood that, but I have no idea/experience what "flows over that line" :-)
I googled around a bit but was only able to find question on that and people ended up on infrared :-(

So I have no idea :-(

Apollon77, 2018-05-18:
Please reopen when you have more infos

closed 2018-05-18 by Apollon77