Apollon77/smartmeter-obis #38: Uncaptured closure in SerialResponseTransport

toarnold, 2018-11-11:
Form time to time I get an error in line 223 of SerialResponseTransport.js. This might be caused by an issue in the underlying serial library. But this error will not be handled correctly:

`
         if (this.serialComm) this.serialComm.close(function(err) {
            if (err) {
                this.options.logger('ERROR CLOSING SERIALPORT');
            }
        });
`

`this.options` didn't exist in case of invoking the callback function. It has to be captured in a closure before.

Do you accept a PR?

Apollon77, 2018-11-11:
Sure. PRs are welcome ;-) I will also take a look later ...

Apollon77, 2018-11-11:
I do not nee a PR ... fixed it hopefully. WIll publish new version soon

closed 2018-11-11 by Apollon77

Apollon77, 2018-11-11:
1.1.3 on npm

commit a3f8dae2 (2018-11-11): fixes #38