Apollon77/smartmeter-obis #54: Telnet connection

oresch, 2019-04-11:
Hi there,

via the ioBroker website I came to this site and saw, that there is only a serial or http connection available to your adapter. I own a serial-ethernet adapter where I can connect via telnet:5000 to it and get the SML message out of the sml adapter.
How can I manage now to connect to the the sml adapter with our ioBroker adapter?
thanks
Oliver

Apollon77, 2019-04-11:
What is the exact protocol there? Is it just connection on TCP raw level and then data is coming?
It would be needed to enhance the library comparable to SerialResponseTransport with a TCPResponseTransport ... it dshould be relatively straight forward to write it ... just replace all the Serial stuff with TCP stuff :-)

PRs are welcome

oresch, 2019-04-11:
Hi,
here a printout. The device sends such datagram every second or two:

```
telnet 10.76.78.104 5000
Trying 10.76.78.104...
Connected to 10.76.78.104.
Escape character is '^]'.
mac=00:08:DC:XX:XX:XX
boot=1
fw=02.00
time=1260424760
zpb=DE0000000000000000000000000000001
SML(1B1B1B1B01010101760503B015EC620062007263010176010105013AB1FA0B090149534B00043744F8010163DB8100760503B015ED620062007263070177010B090149534B00043744F8070100620AFFFF7262016501BBF7507A77078181C78203FF010101010449534B0177070100000009FF010101010B090149534B00043744F80177070100010800FF650000018201621E52FF59000000000266B8D30177070100010801FF0101621E52FF59000000000266B8D30177070100010802FF0101621E52FF5900000000000000000177070100100700FF0101621B520055000000E20177070100240700FF0101621B5200550000005E0177070100380700FF0101621B5200550000004E01770701004C0700FF0101621B520055000000350177078181C78205FF010101018302B5882C4C38F95514DBB2D605B8DD7E56170C94E454EDACB0123CFE281D9C4984892C63E3F6F4D784B4907F582F1DB6BA01010163371800760503B015EE620062007263020171016345D3001B1B1B1B1A001956)
mac=00:08:DC:XX:XX:XX
boot=1
fw=02.00
time=1260424761
zpb=DE0000000000000000000000000000001
SML(1B1B1B1B01010101760503B015EF620062007263010176010105013AB1FB0B090149534B00043744F8010163249E00760503B015F0620062007263070177010B090149534B00043744F8070100620AFFFF7262016501BBF7527A77078181C78203FF010101010449534B0177070100000009FF010101010B090149534B00043744F80177070100010800FF650000018201621E52FF59000000000266B8D40177070100010801FF0101621E52FF59000000000266B8D40177070100010802FF0101621E52FF5900000000000000000177070100100700FF0101621B520055000000E30177070100240700FF0101621B5200550000005E0177070100380700FF0101621B5200550000004E01770701004C0700FF0101621B520055000000350177078181C78205FF010101018302B5882C4C38F95514DBB2D605B8DD7E56170C94E454EDACB0123CFE281D9C4984892C63E3F6F4D784B4907F582F1DB6BA01010163ECBD00760503B015F162006200726302017101634CBE001B1B1B1B1A000D95)
```

oresch, 2019-04-12:
As I'm not that JS programmer the result will get into a mess ;-) So maybe you'll wanna add it this request on your roadmap?

Apollon77, 2019-04-12:
Issue is already flagged ;-)

Apollon77, 2019-06-25:
maybe https://github.com/chris1705/smartmeter-obis/commit/19da11a7b02838c2e4a306b911cc6662f1c560c1 could be a base :-)

Apollon77, 2020-03-08:
done with smartmeter-obis 2.1.0 (experimental because not self tested, so feedback required)

closed 2020-03-08 by Apollon77