Apollon77/smartmeter-obis #63: Error while parsing SML message: Error: Wrong TL-Field 0x65 for Choice!

bluesky81, 2019-08-03:
Hello!
I get the Error Wrong TL-Field 0x65 for Choice.

For me the message looks good.
Could you please verify what could be wrong? :-)

1b1b1b1b010101017604000001620062007265000001017601010700000052fcd90b0a01484c59020002015a0101630d990076040000026200620072650000070177010b0a01484c59020002015a01017777070100603201010101010104484c590177070100600100ff010101010b0a01484c59020002015a0177070100010800ff65001ca904650052fcd9621e52ff65002f213c0177070100020800ff65001ca904650052fcd9621e52ff65000dd4b70177070100000200000101010109312e30322e3030370177070100605a02010101010105413031410177070100600500ff0101010165001ca90401010163a3d200760400000362006200726500000201710163e82300001b1b1b1b1a016868

Apollon77, 2019-08-03:
Can you find a SML version number in the specs of the smart meter?

bluesky81, 2019-08-03:
It is SML-Specification-Version 1.04...

Apollon77, 2019-08-03:
And the error and possition is really the same in all cases? As soon as it is not the chance of being a transfer error is much higher!

bluesky81, 2019-08-03:
Yes, with every cycle comes the same error...
Are there further analysis possibilities?

Apollon77, 2019-08-04:
I now need to check the message and the specs ... we will see.

Which adpter version do you use? (Just to make sure)

Apollon77, 2019-08-05:
Ok, what exact smart meter is it?

I checked it and the message is incorrect. There are 4 bytes missing in two values inside, so in fact  it more seems to be a implementation error in that smart meter then a reading issue.

Best idea would be to contact your  energy company and tell about the wrong SML content. Provide them with the message above and tell them:
- In the Value list the items 3 and 4 are missing the "Choice" data for the value timestamp according to the SML specifications

If I insert those values manually I can completely parse the message - only the CRC16 checksum is then invalid (which is normal because I modified it).

bluesky81, 2019-08-08:
Hi,
thank you for your assistance!
Sorry, for my late reply, I was on vacation.
I think the specification says that actSensorTime (SML_Time) is OPTIONAL?
So it doesn't have to be implemented, right?

Apollon77, 2019-08-08:
The topic is that the Time-Value is present, but the "Choice" before that defines the "format" of the timestamo is missing:

>Das Element ‚SML_Time‘ wird entweder als Sekunden-Index oder als Zeitstempel angegeben.
(F) SML_Time {
secIndex timestamp
::= CHOICE
[0x01] Unsigned32, [0x02] SML_Timestamp
}

"OPTIONAL" means in this case that it can be "01" in data stream if it is not present, but also this is not existing ... I can only see a invalid protocol implementation and also see no real way to get around that automatically.

So in fact I was able to parse your message by adding 4 bytes (the Choice field identifier and the choice number 01/02) in the said two places and everything was correct.

hacki11, 2019-10-31:
https://github.com/Apollon77/OpenSML/pull/1

Apollon77, 2019-11-25:
will be fixed in next version

closed 2019-11-25 by Apollon77