Apollon77/smartmeter-obis #75: DZG DVS74 fehlerhafte Firmware

Johannes1116, 2019-12-22:
Hallo ich möchte über node red mein Stromzähler auswerten. Ich vermute das ich von der fehlerhaften Firmware betreffen bin. (Hier beschrieben https://wiki.volkszaehler.org/hardware/channels/meters/power/edl-ehz/dzg_dvs74) lässt sich der Workaround auch hier durchführen?
Grüße Johannes

Apollon77, 2019-12-22:
Was genau passiert denn? Gibts nen Fehler wen du die Library nutzt? Welchen? Wie sieht ein vollständiges sml aus?

Oder ist nur der Wert der rauskommt falsch?

Mehr Details bitte.

Johannes1116, 2019-12-22:
Der Wert vom aktuellen Verbrauch wird meistens als irgendein Negativer Wert angezeigt ab und an auch als positiver aber auch nicht stimmender Wert. Habe es schon mit den verschiedensten System versucht war immer das gleiche Ergebnis. Kam dann nach langem Suchen auf die Seite von Volkszähler.

Apollon77, 2019-12-22:
Naja dann ist die einfachste Lösung das Du den Wert nachträglich umrechnest. Wenn DU den Fehler kennst  kannst Du ihn doch auch umrechnen?

Johannes1116, 2019-12-23:
Die Werte werden falsch interpretiert dadurch kann ich sie nicht umrechnen. Sind ja auch mal positiv dann wieder negativ.

Apollon77, 2019-12-23:
Was verstehst Du unter interpretiert? Der Wert wird aus dem Datenstrom ausgelesen als Signed Wert mit x Bits. Bedeutet wenn Du einen Buffer erzeugst der lang genug ist, den Wert wieder als Signed reinschreibst und als unsigned ausliesst solltest Du exakt haben was Du willst. Das mit dem Buffer ist definitiv der schwerere Weg, es geht garantiert auch anders.

Auf der Binärebene sind beide Zahlen das gleiche ...

Johannes1116, 2019-12-23:
Aus Zug aus der Volkszählereite "Achtung: Zumindest einige Versionen der SML-Firmware haben einen Fehler in der Kodierung positiver Verbrauchswerte, diese werden falsch als „signed int“ kodiert und somit falsch interpretiert, ein workaround ist hier zu finden. Wir sind auch diesbezüglich in Kontakt mit dem Hersteller." das meinte ich mit falsch interpretiert. Dort wird als Workaround empfohlen die sml_server.c abzuändern.

Apollon77, 2019-12-23:
Am bedenkst die Aussage ja korrekt. Wenn die Zahl als signed gelesen wird aber unsigned Enthalten sind ist das quasi eine falsche Interpretation. 

In meiner Welt lässt sich das aber umrechnen. 

Sag doch mal einen falsch gelesenen Wert und wie der korrekte wäre.

Johannes1116, 2019-12-23:
Ausgelesen -161.43 auf Stromzähler 496 ab und zu kommt kein Negativer Wert wie muss ich das dann umrechnen?

Apollon77, 2020-02-03:
Sorry das es so lang gedauert hat. Umrechnung mit

const richtig = ((falsch*100) &0xffff) / 100;

In dem Fall wird aus -161,43 ein 493,93

closed 2020-02-03 by Apollon77

Apollon77, 2020-03-08:
hats geklappt?

Johannes1116, 2020-03-08:
Habe versucht die Formel in eine Funktionnode zu schreiben das hat leider nicht auf Anhieb geklappt. Hatte mich dann erst mal nicht mehr damit beschäftigt.

Apollon77, 2020-03-08:
Für weitere Hilfe mach doch mal nen Thread im Forum auf.

Mugridge, 2022-05-11:
> 
Oder einfach den negativen Wert von 655.36 subtrahieren.

mopkick, 2022-05-27:
> > 
> 
> Oder einfach den negativen Wert von 655.36 subtrahieren.

Wie genau kann man den Wert subtrahieren? Hat das schon jemand "gebaut"?

Apollon77, 2022-05-27:
@mopkick was genau ist die Frage? JavaScript Code ist weiter oben bzw `const richtig = 655.36 + falsch;`ist scheinbar andere variante ohne es jetzt nachgerechnet zu haben ...

mopkick, 2022-05-27:
> @mopkick was genau ist die Frage? JavaScript Code ist weiter oben bzw `const richtig = 655.36 + falsch;`ist scheinbar andere variante ohne es jetzt nachgerechnet zu haben ...

Bin da nicht so fit in dem Thema haben nur das gleiche Problem mit dem DVS74. Nutze den Standard iobroker mit Smartmeter. Wenn ich wüsste wie ich den die Berechnung anpassen kann wäre mir schon geholfen.

Apollon77, 2022-05-27:
Dann würde ich an Deiner Stelle mal im ioBroker Forum fragen mit verweis hierher. Am Ende brauchst Du ein kleines JavaScript scriptchen was den falschen wert bekommt und den richtigen Wert in einen anderen State schreibt

mopkick, 2022-05-27:
Danke für die Info 👍 
Eventuell ist es auch über die Influxdb / Grafana realisierbar die den Wert ausbügelt. Da schau ich mal was da möglich.