Apollon77/smartmeter-obis #106: add libsml-testing for regression testing and

devZer0, 2020-03-28:
hi, what about adding https://github.com/devZer0/libsml-testing to the tests?

 i.e. 

- make an example how to download those binary dumps and decode them with smartmeter-obis
- make such test part of the "testing procedures" to test for regression

Apollon77, 2020-03-28:
On one hand I like that idea. On the other hand this should be more included in the testing from the nodejs sml lib which is used, or?! The library here relies for sml parsing that und used lib works and the lib has detailed tests already for all the features the library provides.
So to include Libsml tests here feels not the correct place. 

Do we should move that issue to OpenSML lib. What do you think?

devZer0, 2020-03-28:
yes, you are absolutely right, it wasn't clear to me that there is a separate git repo for this - i guess: https://github.com/Apollon77/OpenSML , correct ?

Apollon77, 2020-03-28:
Yes but open in mail repo. The one in my space is just a fork. Open with main repo please. I work in that too ;-)

devZer0, 2020-03-28:
done

closed 2020-03-28 by devZer0