Apollon77/smartmeter-obis #124: SML message parsing errors

MichaelAnders, 2020-08-12:
I own a two-way smart meter (Iskra eHZ-EDL Type MT681-D4A52-KOp) for my solar panels. Reading the data from the meter works fine using the 47_OBIS module in FHEM but fails in Node-Red using node-red-contrib-smartmeter (which uses this code).

Node-RED 1.1.3
nodejs 12.18.3
npm 6.14.6

The following are error messages I get:

smartmeter-obis error: Error: Error while parsing SML message: Error: Unknown TL-Field for SmlListEntry!: Message:
1b1b1b1b010101017605173f87d162006200726301017601010507bfd7ef0b0649534b010722f8c424010163aef9007605173f87d2620062007263070177010b0649534b010722f8c424070100620affff726201650b35ba3b7a77078181c78203ff010101010449534b0177070100000009ff010101010b0649534b010722f8c4240177070100010800ff65000101a201621e52ff5900000000094fd7280177070100010801ff0101621e52ff59000101621e52ff5900000000000000000177070100020800ff65000101a201620100020801ff0101621e52ff590000000010506cfd0177070100020802ff0101621e52ff59000000000000000001fffff5fb0177078181c78205ff0101010183021278b1e04c3d323e3799add9fd6ebd6c9cc75ced19b6db168397c4a2e79c4073c601010163e3d600760517001b1b1b1b1a0049b1

smartmeter-obis error: Error: Error while parsing SML message: Error: Wrong TL-Field 0x49 for Choice!: Message:
1b1b1b1b010101017605173f88010b0649534b010722f8c424010163dbe6007605173f8802620062007263070100620affff726201650b35ba4d7a77078181c78203ff010101010449534b0177070100000009ff010101010b0649534b010722f8c4240177070100010800094fd7280177070100010801ff0101621e52ff5900000000094fd72801000000000000000177070100020800ff65000101a201621e52ff5900000000621e52ff590000000010506d7d0177070100020802ff0101621e52ff5900000101621b520055fffff5ec0177078181c78205ff0101010183021278b1e04c30eda67145e343fd6ebd6c9cc75ced19b6db168397c4a2e79c4073c60101016302017101638f3e001b1b1b1b1a00b38e

smartmeter-obis error: Error: Error while parsing SML message: Error: Wrong TL-Field (1) for Unsigned!: Message: 
1b1b1b1b010101017605173f881c62006200726301017601010507bfd8080b0649534b010722f8c4240101639432007605173f881d620062007263070177010b0649534b010722f8c424070100620affff726201650b35ba577a77078181c78203ff010101010449534b0177070100000009ff010101010b0649534b010722f8c4240177070100010800ff65000101a201621e52ff5900000000094fd7280177070100010801ff0101621e52ff5900000000094fd7280177070100010802ff0101621e52ff59000000000000000001770701000208000010506dc50177070100020801ff0101621e52ff590000000010506dc50177000000000000000177070100100700ff0101621b520055fffff5ea017707814c3d323e3799add92a16fdcf9369eaf830eda67145e343fd6ebd6c9cc75ced19b6db168397c4a2e79c4073c6010101630201710163a4f8001b1b1b1b1a0050e3

smartmeter-obis error: Error: Error while parsing SML message: Error: Wrong TL-Field 0x53 for Choice!: Message:
1b1b1b1b010101017605173f883762006200726301017601010507bfd8110b0649534b010722f8c4240101637225007605173f8838620062007263070177010b0649534b010722f8c424070100620affff726201650b35ba617a77078181c782000009ff010101010b0649534b010722f8c4240177070100010800ff650001280177070100010801ff0101621e52ff5900000000094fd7280177070100010000000177070100020800ff65000101a201621e52ff590000000010506e0d590000000010506e0d0177070100020802ff0101621e52ff590000000000000055fffff5dd0177078181c78205ff0101010183021278b1e04c3d323e37e343fd6ebd6c9cc75ced19b6db168397c4a2e79c4073c601010163d16a007605173f8839620062007263020171011b1b1b1b01010101

smartmeter-obis error: Error: Error while parsing SML message: Error: Wrong TL-Field 0x49 for Choice!: Message:
1b1b1b1b01010101760507bfd81b0b0649534b010722f8c4240101634f34007605173f88566200f8c424070100620affff726201650b35ba6c7a77078181c78203ff0101010101010b0649534b010722f8c4240177070100010800ff65000101a201621e52ff5900000000094fd7280177070100094fd7280177070100010802ff0101621e52ff5900000000000000000177070100020800ff65000101a201621e520801ff0101621e52ff590000000010506e5d0177070100020802ff0101621e00100700ff0101621b520055fffff5ec0177078181c78205ff010101018302cf9369eaf830eda67145e343fd6ebd6c9cc75ced19b6db168397c4a2e79c4062006200726302017101639adc001b1b1b1b1a002251

smartmeter-obis error: Error: Error while parsing SML message: Error: Unknown TL-Field for SmlListEntry!: Message: 1b1b1b1b010101017605173fb81662006200726301017601010507bfe8060b0649534b010722f8c4240101635c26007605173fb817620062007263070177010b0649534b010722f8c424070100620affff726201650b35cbee7a77078181c78203ff010101010449534b0177070100000009ff010101010b0649534b0107220177070100010801ff0101621e52ff5900000000094fd7760177070100010802ff0101621e52ff5900000000000001621e52ff59000000001050cfda0177070100020801ff0101621e52ff59000000001050cfda0177070100020802ff0101621e52ff5900000000000000000177070100100700ff0101621b520055fffffe630177078181c78205ff0101010183021278b1e04c3d323e3799c4a2e79c4073c6010101634fcc0076bb33001b1b1b1b1a003efa

Apollon77, 2020-08-12:
I think the difference is that fhem only reads special values and we here really parse the sml. 

Do these error always come or do it also work from time to time?

With this what you describe please make sure the head is positioned correctly, that the cable is good grounded, that USB port provides enough power, pot try an active usb hub if you directly have put it in a computer. Also try to hold cable best possible separated from the power cables. 

Is it getting better then? In fact it sounds like data transfer problems because the error is always in different places in the message.

MichaelAnders, 2020-08-14:
Hardly any data gets through, normally I only get error messages like the one above.

System is an ATX with Ryzen CPU and B450 chipset - USB ports should be sufficient with power I assume.

I have fhem and node-red running in the same Proxmox VM, and I just enable/disable the device accordingly in either one.

In fhem I always get results - every second an update.

So I am not sure if this is really a hardware/cable issue (I did move the cable away from the power cables though)...

Apollon77, 2020-08-14:
As described: FHEM ist just searching for certain parts of the SML message, so if there is no data error in this then it will "always work". This library parses the SML data message

So please do what I told above - I also had one of my IR head on a Intel NUC USB port and now added an active USB hub because it was in fact partly not enough for the cable length.

And also as said: because of the fact that the error always seems to be on a different place it is most likely no firmware issue where the device sends a broken message, so it is from my experience most likely that data are corrupted.

So again:
* Please check the position of the head (it could be a millimeter ...)
* Please try with an active USB hub

closed 2020-11-19 by stale[bot]