Apollon77/smartmeter-obis #139: Replace dependency `request`

pkwagner, 2020-12-06:
Installing `smartmeter-obis` throws following warning:

```
warning meter2mqtt > smartmeter-obis > request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
warning meter2mqtt > smartmeter-obis > request > har-validator@5.1.5: this library is no longer supported
```

There's also an official deprecation announcement on their [GitHub repo](https://github.com/request/request), hence finding an alternative library might be a good idea in the medium run.

Apollon77, 2020-12-06:
Yes known, but works still well andno other features needed, So if someone has time before me I'm happy to accept a PR :-)

robertoschwald, 2025-08-31:
I replaced request with axios locally. IMHO it would be better to use fetch directly, so do not rely on another dependency. Is it planned to stick with node >= 16? If not, I provide a PR which lifts it to 18 (or better: 20, because 18 is deprecated already) and uses fetch.

Apollon77, 2025-09-14:
@robertoschwald I am fine to go with node.js 18+ and replace request by fetch directly

Apollon77, 2025-09-14:
BTW: I can also ask copilot to do it ... let me try :-)