Apollon77/smartmeter-obis #156: smartmeter-obis error: Error: No or too long answer from Serial Device...

JE3001, 2021-04-29:
I'm not sure if this is the right forum, however I hope somebody can help me to get data to node red from my smart electrical power meter.
I've issued already the problem here...
https://github.com/coolchip/node-red-contrib-smartmeter/issues/28#issue-869159150

However, when I try to get data from the smartmeter I only receive the error massage
**"smartmeter-obis error: Error: No or too long answer from Serial Device after last request."**

What is wrong in my settings? Can anybody help?

Thanks and Regards!

Apollon77, 2021-04-29:
Provide full debug level logging please. I have no idea how to activate that in node-red. Main question is if data are coming in or not at all and if coming in if settings are correct

JE3001, 2021-04-30:
Thanks for your response!

Also in the red node logging I receive only the above mentioned error massage (nothing else related to the serial port / smartmeter node). 

Node Red Debugger:
[image]

Node Red Logging (5th line):
[image]

I've checked the serial port via Minicom...
[image]
 
For me it looks some data coming but maybe I'm wrong.

Any hint for me what else can be checked?

Apollon77, 2021-04-30:
Ok, I can only speak for the library: there is a loglevel that can be set to get more logs ... no idea how you can enable that in node-red-plugin.

Yes it seems that "some data" coming in there, but without a hey dump of it I can not say anything about if data lokks SML-ish or not

JE3001, 2021-05-01:
Ok, I will ask in the Node Red forum if it is possible to change the  loglevel of the node

closed 2021-05-09 by Apollon77

coolchip, 2021-12-17:
I've added a config parameter "debugging" to the node-red node. This enables everybode to put smartmeter-obis in debug mode. Then you are able to copy hex dumps.