Apollon77/smartmeter-obis #203: Measure values at fixed times instead of a rolling interval

MarkusSchneider, 2022-03-03:
**Is your feature request related to a problem? Please describe.**
I like to have the values measured at a fix timestamp e.g every full hour or every 5 minutes.
In the moment it is possible to define a requestInterval. 
But the exact time, when a measurement takes place, depends on the boot time.

**Describe the solution you'd like**
I like to have the measurements collected at fixed points in time not only at fixed intervals.
setTimeout should take the current time in account and calculate a timeout to match the full requestInterval.
https://github.com/Apollon77/smartmeter-obis/blob/b1dc555883085f4d17dcf55b9c57c3b3f940f721/lib/transports/SerialResponseTransport.js#L219-L224

e.g. requestInterval = 300 seconds. 
Current Time 1:02:32, next valid execution time is 1:05:00. 
Wait for 1:05:00  - 1:02:32 seconds = 148 Seconds.

**Describe alternatives you've considered**
Startup could be deferred but has the disadvantage that the measurement time will drift over time because communication to smart meter and processing of data will take time and values are no longer collected to defined time stamps.

**Additional context**
If you like this feature, let me know if I can help. I can file a pull request.

Apollon77, 2022-03-03:
In fcat it will never be "accurately on this timestamp" because communicstion and parsing will take time - especially when you have a smartmeter that sends data from "himself" then how do you know that a data packet is sent "exactly at the time you like it"? In my eyes this will never be "exact". That would be a "promise to the users" that the adapter can never fulfiill.

Honestly: If you have a smartmeter that sends the data from himself simple remove the request interval (set to 0) and let the adapter parse any data packet - then you can have an extra script which selects the "latest data" on the timepoints you like.

This is my opinion about that ... But we can leave it open here so that other users can vote.

MarkusSchneider, 2022-03-03:
Ok, I can handle this in the database. From my point of view it is not import that the reading is done exactly to the requested timestamp. 
Important is that the reading timestamp is equidistant (took place at a multiple of the requestInterval) so I can calculate e.g. the consumption of an hour or day.

Apollon77, 2022-03-03:
> Important is that the reading timestamp is equidistant (took place at a multiple of the requestInterval) 

But why then not setting interval to 0 and read simply all values ... that you always can select the value which is closest to the timepoint you need.

MarkusSchneider, 2022-03-03:
Yes setting interval to 0 or 1 is a valid option. I can then pick the required readings afterwards from the database. E.G running a task every 5 min that took the reding best matching to the required time stamp.

I'll see what I can do with InfluxDB 😊.

Thank you for your support.

closed 2022-03-03 by MarkusSchneider