Apollon77/smartmeter-obis #210: Reading values from DVS74 DZG is wrong sometimes

cracker0dks, 2022-05-10:
I get correct values but then some wrong values, until I get correct values again :(

**To Reproduce**  
Steps to reproduce the behavior:
Im Reading with the following config:
```
var options = {
    'protocol': "SmlProtocol",
    'transport': "SerialRequestResponseTransport",
    'transportSerialPort': "/dev/ttyUSB1",
    'transportSerialBaudrate': 9600,
    'requestInterval': 10,
    'obisNameLanguage': 'de',
    'debug': 0,
    'protocolSmlIgnoreInvalidCRC' : true
};
```
And getting this data:
[image]

The first values are correct, then we go down (Big first step down), until we get back up (with a big step) to the correct value again... and go down to an incorrect value again at the end.

I saw the bug here: https://github.com/Apollon77/smartmeter-obis/issues/75 
this might be the same problem but with no solution so I add more data to might fix the problem.

What is interesting about this -> the big steps with the following values:
383 to 256
383 to 512
639 to 512
So as you might notice that the steps always ends on a number with powers of two and the difference is always 127 (128 but minus one because we added one). This can not be a coincidence if you ask me ;)

Do we have any  way to get the correct values out of that? Thanks for any hint

Apollon77, 2022-05-10:
If the values come out from SML that way then in first run it is that way. You can send/attach here a debug log with the full message and the parsed SML values and tell which value it is exactly and I can re-check the decoding of the message to make sure it is not a bug in the SML decoding (I would consider it uncommon, but sometimes you never know)

if it is a firmware buig writng invalid valus then I have no chance ... If you know how to fix it then you can write a small javascript to get a fixed value for yourself

cracker0dks, 2022-05-10:
This is the output:
[out.txt](https://github.com/Apollon77/smartmeter-obis/files/8661711/out.txt)

1-0:2.8.0*127 is 530, but should be 659
thanks

Apollon77, 2022-05-11:
Then it really seems to be the Firmware bug because dattypes in both are the same .... I would not see any library logic issues

closed 2022-05-11 by Apollon77