Apollon77/smartmeter-obis #218: Add Obis String 1-1:30D, Obis String 1-1:7D, Obis String 1-1:1D

joerghildering, 2022-09-16:
I´m using the **ELSTER 1440** and the smartmeter adapter in iobroker an got this in the protocol:

Error while parsing D0 content: ignore complete part Error: Invalid Obis String **1-1:30D**
Error while parsing D0 content: ignore complete part Error: Invalid Obis String **1-1:7D**
Error while parsing D0 content: ignore complete part Error: Invalid Obis String **1-1:1D**

Could you please add this strings?

Apollon77, 2022-09-16:
Honestly ... unsure ... The OBIS specs is not defining such IDs ... Whats the meaning of them? Seems like the manufactorer invested them himself :-( Can I please get a debug lkog to see the message returned? Is this coming anytime data coming in or just sometimes?

joerghildering, 2022-09-16:
Is this coming anytime data coming in. I contacted the manufacturer for this Info and for now I'm waiting for a feedback.
I only got back this in iobroker debug log:

smartmeter.0 | 2022-09-16 15:24:54.470 | info | Received 257 values, 7 updated
-- | -- | -- | --
smartmeter.0 | 2022-09-16 15:24:54.334 | warn | Error while parsing D0 content: ignore complete part Error: Invalid Obis String 1-1:30D
smartmeter.0 | 2022-09-16 15:24:54.334 | info | Error: Error while parsing D0 content: ignore complete part Error: Invalid Obis String 1-1:30D
smartmeter.0 | 2022-09-16 15:24:54.334 | warn | Error while parsing D0 content: ignore complete part Error: Invalid Obis String 1-1:7D
smartmeter.0 | 2022-09-16 15:24:54.333 | info | Error: Error while parsing D0 content: ignore complete part Error: Invalid Obis String 1-1:7D
smartmeter.0 | 2022-09-16 15:24:54.332 | warn | Error while parsing D0 content: ignore complete part Error: Invalid Obis String 1-1:1D
smartmeter.0 | 2022-09-16 15:24:54.332 | info | Error: Error while parsing D0 content: ignore complete part Error: Invalid Obis String 1-1:1D

Apollon77, 2022-09-16:
Ok, if it comes always then it seems wrong in response ... but as said formally not defined for OBIS :-( Lets see what they answer

Apollon77, 2022-12-16:
@joerghildering Aand, any answer from manifacturer?

joerghildering, 2022-12-16:
@Apollon77 Unfortunately not