Apollon77/smartmeter-obis #219: Output direction of energy flow

StofLE, 2022-09-23:
Direction of energy flow is shown in meter display as **-A** or **+A**.

There is an additional bit in 1.8.0 to see it in the data. Found this information here:
[https://www.mail-archive.com/volkszaehler-dev@demo.volkszaehler.org/msg02161.html](url)
[https://wiki.volkszaehler.org/hardware/channels/meters/power/edl-ehz/emh-ehz-h1](url)

In my case
Exporting energy (status **182**):
```
Sep 23 13:12:07 zaehler Node-RED[179291]: #011Obj-Name: 1-0:1.8.0*255
Sep 23 13:12:07 zaehler Node-RED[179291]: #011Status: 182
Sep 23 13:12:07 zaehler Node-RED[179291]: #011Val-Time:
Sep 23 13:12:07 zaehler Node-RED[179291]: #011Unit: 30
Sep 23 13:12:07 zaehler Node-RED[179291]: #011Scaler: -1
Sep 23 13:12:07 zaehler Node-RED[179291]: #011Value: 896203604
Sep 23 13:12:07 zaehler Node-RED[179291]: #011Value-Signature: undefined
```

Importing energy (status **1b0**):
```
Sep 23 13:12:40 zaehler Node-RED[179291]: #011Obj-Name: 1-0:1.8.0*255
Sep 23 13:12:40 zaehler Node-RED[179291]: #011Status: 1b0
Sep 23 13:12:40 zaehler Node-RED[179291]: #011Val-Time:
Sep 23 13:12:40 zaehler Node-RED[179291]: #011Unit: 30
Sep 23 13:12:40 zaehler Node-RED[179291]: #011Scaler: -1
Sep 23 13:12:40 zaehler Node-RED[179291]: #011Value: 896203650
Sep 23 13:12:40 zaehler Node-RED[179291]: #011Value-Signature: undefined
```

Would be cool to see this in the output for processing it in home automation systems...

Stefan

Apollon77, 2022-09-23:
"duplicate" of https://github.com/Apollon77/ioBroker.smartmeter/issues/306

When you use the library you shpuld have the data available already .. just parse it :-)) The question that I do not know if  this is standardized how and where it is encoded how. I would not wonder igf this is highly manufactorer specific ...

So Library wise the Status field and it's value is there ... so you it would up to you to interpret it according to the specs of your meter?!

StofLE, 2022-09-23:
Thanks for your answer.

Honestly said I do not use the library directly. I readout a german meter EMH eHZ with **Node Red** and another github project  (https://github.com/coolchip/node-red-contrib-smartmeter). And this guy uses your library. But i only can enable debug to the system console;

Stefan

Apollon77, 2022-09-23:
Then maybe open an issue there to make the status field available so that you can use it in your flow

Tinbo888, 2023-04-16:
I have the same error in debug mode i see the Status 182 when i buy energy or 1a2 when i sold energy,
but there is no status in the object explorer only the counter value so i cant figure out is the power value
positiv or negative. In the moment it is for me only possible to set the adapter to debug, parse the debug log
with the logparser adapter to fit my needs, many tons of garbage spams my logs so it can not be the right way.
Hopefully it destroys not my sd card.
In the screenshot below at the red arrow i wish to see the value "182" or "1a2"
[image]
the red arrow value "182" or "1a2"  im missing
[image]
here is nothing to see with "182" or "1a2"  the  1-0:15_7_0__255.value is always positive
[image]