Apollon77/smartmeter-obis #220: Hichi IR is not connection with the smartmeter instance

FenziRocks, 2022-09-28:
**Describe the bug**  
The Smartmeter is not connection with my Hichi IR.

**To Reproduce**  
Steps to reproduce the behavior:
1. have mac os with usb C
2. go to smartmeter setting and chose your serial connection
3. the instance still say that is is not connected
4. and the IO broker log shoots the error "No or too long answer from Serial Device after last request."

**Expected behavior**  
The device can be connected with smartmeter
The error message comes from smartmeter-obis repo and not from iobroker, thats why I'm asking here

**Screenshots & Logfiles**  
[image]
[image]
[image]

**Versions:**  
 - Adapter version: 3.3.4
 - JS-Controller version: 4.0.23
 - Node version: v16.17.1
 - Operating system: MacOS Monterey 12.5.1
 - Browser: Chrome Version 105.0.5195.125 (Offizieller Build) (arm64)

**Additional context**  
I have tested my Hichi IR with Hterm on a windows  and it works.
Unfortunately I also have some other issues on the windows machine with the connection where it not even load the the serial port list -> but this is another ticket maybe

Apollon77, 2022-09-28:
Pleaseopen the issue in iobroker.smartmeter GitHub repo because it is about the adapter. Additionally Provide a screenshot of the settings and provide a Debug log. The error just states that the device is not responding, so it can be anything starting with  settinngs. SO please also provide what devcie we talk abot? Do you get results when asccessing the serial device directly?

maybe also consider to ask in ioBroker Forum in the smartmeter adapter thread because most likely it is a configuration/usage issue then a actual problem

closed 2022-09-28 by Apollon77

FenziRocks, 2022-09-28:
Thanks for reaching out so quick.
As the error message comes from your repo I thought it could be a topic for it.
https://github.com/Apollon77/smartmeter-obis/blob/7221fbd200b88d88bbe458db4f4d9fea796d6d27/lib/transports/SerialResponseTransport.js#L291

[image]

Apollon77, 2022-09-28:
Why you use "D0 mode override? Pleas epost a debug log. And yes the error comes from the library because there was no message received in the timeout you defined. So all questions above stay relevant