Apollon77/smartmeter-obis #247: Constantly get the error message: "Error: Error on Baudrate changeover: Error: Port is not open"

brand1011, 2024-07-11:
**Describe the bug**  
 I use this librarty through the adapter from NodeRed "node-red-contrib-smartmeter".
After setup a reader in a nodered flow, i get the following error message in a short intervall: "Error: Error on Baudrate changeover: Error: Port is not open" . This error message comes from this library.

**To Reproduce**  
Steps to reproduce the behavior:
1. Setup a IR reader connected to /dev/ttyUSB0
2. configure a baud rate of 300, with 7E1
3. start the metering...

**Expected behavior**  
Receiving some kind of data

**Versions:**  
 - Adapter version: node-red-contrib-smartmeter Version 0.6.3 (unfortunately its 2 years old).
 - JS-Controller version: dont know. i am using nodered
 - Node version: v20.15.1
 - Operating system: debian

**Additional context**  
Add any other context about the problem here.

Apollon77, 2024-07-11:
This error happens when the D0 protocol wants to to a baudrate change but the Serial port is not opened, so it can not do this.

If this happens something is wrong on how this is integrated or such.

ideally i would need to get a debug log of the library ... but maybe better to open an issue with the node-red node repo so that this developer can look into it too?

brand1011, 2024-07-13:
I know the meaning of the error message, but the port is definitely open. When i use vzlogger as an alternative, everything works fine. But i would prefer to use nodered to read out my smartmeters.
I have opened an issue at the nodered node too, but it seems it is not developed further anymore. The latest version is over 2 years old :(

Apollon77, 2024-07-15:
Because you are the only one reporting this I can only guess. An in fact "port open "means that the nodejs process opened it. So try to find out please how to get extended debug logs from the library. I do not now the nodered node so I can not help here.

brand1011, 2024-07-16:
Ich versuchs mal auf deutsch, weil mein englisch nicht so perfekt ist. Sorry.
Ich denke es versuchen nicht viele die Kombination aus Nodered in docker und dem Smartmeter Node. 
Ich hab mir den Source von dem node-red-contrib-smartmeter angeschaut und darin steckt keine Programmlogik. Es ist im Prinzip nur ein UI Wrapper um deine Lib in NoderRed einzubinden und mit den nötigen Parametern zu füttern.
Man sieht auch, dass die Fehlermeldung "Port is not open" aus Deiner Lib, kommt und nicht aus der Node.
Ich versuche mal ein ausführlicheres Debug Log zu bekommen. Danke Dir schonmal, dass Du überhaupt hier antwortest :)

brand1011, 2024-07-16:
Hier die Debug Ausgabe:

`STOP
CLEAR MESSAGE TIMEOUT TIMER
CURRENT SIGNON MESSAGE COUNTER 0 OF 1
DELETE rest of data (0) because new Message start with a new Request Message
SCHEDULE NEXT RUN IN 1000s
CREATE SERIALPORT: 300 7 1 even
SERIALPORT RESET BAUDRATE TO 300

Error: Error on Baudrate changeover: Error: Port is not open
    at SerialPort.<anonymous> (/data/node_modules/smartmeter-obis/lib/transports/SerialRequestResponseTransport.js:287:52)
    at SerialPort._error (/data/node_modules/@serialport/stream/lib/index.js:198:14)
    at /data/node_modules/@serialport/stream/lib/index.js:205:31
    at process.processTicksAndRejections (node:internal/process/task_queues:77:11)

16 Jul 23:51:44 - [error] smartmeter-obis error: Error: Error on Baudrate changeover: Error: Port is not open
Transport Reset!! Restart = true
STOP`

Apollon77, 2024-07-17:
Naja, code wise ist das aber komisch ...

https://github.com/Apollon77/smartmeter-obis/blob/master/lib/transports/SerialRequestResponseTransport.js#L281-L285

ist die einzige stelle wo das "RESET BAUDRATE" gemacht wird und das ist nachdem der port erfolgreich geöffnet wurde und sdaten geflusht wurden. Also irgendwas passiert da - dneke eher es gibt nen fehler beim open.

Akannst gern bei dem "open" call macjk "function(err)" draus machen und die dann diesen fehler mal loggen lassen.

brand1011, 2024-07-17:
Den Code hatte ich mir nach dem StackTrace auch angeschaut und hab nicht verstanden, warum "open" scheinbar funktioniert, "update" aber nicht. 

> Akannst gern bei dem "open" call macjk "function(err)" draus machen und die dann diesen fehler mal loggen lassen.

DAS war der entscheidende Hinweis!!! Vielen Dank!
Ich habe eine entsprechende Debug Ausgabe eingebaut und heraus kam folgende Meldung:

`SERIALPORT OPEN: Error: Error: Permission denied, cannot open /dev/ttyUSB0`

Da hats geklingelt... Der node-red Container läuft per Default als User "node-red" mit der ID 1000. Dieser darf den USB Port nicht lesen/schreiben. Also habe ich auf dem Host System den entsprechenden User mit ID 1000 der Gruppe "dialout" hinzugefügt und für Docker Compose diese Ergänzung gemacht:

```
    group_add:
      - dialout
```

Jetzt funktioniert alles! Eine andere Lösung wäre wahrscheinlich gewesen den Container als User "root" laufen zu lassen.
Deswegen hatte auch vzlogger funktioniert... Der Container läuft per Default als root.
Vielen Dank für Deine Hilfe! Damit können wir das Ticket schließen.

Apollon77, 2024-07-19:
Ne lass mal offen, ich muss da Error handling und error check einbauen :-)

brand1011, 2024-07-21:
OK :-)

closed 2025-05-06 by stale[bot]