Apollon77/smartmeter-obis #253: ot logging errors in a way that will trigger the catch node

gitisgreat2023, 2025-05-05:
**Is your feature request related to a problem? Please describe.**
Not logging errors in a way that will trigger the catch node

**Describe the solution you'd like**
Thrown errors are able to be catched by the catch node.

**Describe alternatives you've considered**
None

**Additional context**
Add any other context or screenshots about the feature request here.

Apollon77, 2025-05-05:
What exactly is the issue? What you mean? I do not understand anything. Normally all errors should be catched and logged. So please provide a log and exact stack trace and error information.

gitisgreat2023, 2025-05-05:
I have a catch: all active in my flow. When smartmeter-obis throws an error, for example like this: 

[image]

Catch: all doesn't show anything:

[image]

Apollon77, 2025-05-06:
There is no error thrown!

see https://github.com/Apollon77/smartmeter-obis/blob/master/lib/protocols/SmlProtocol.js#L85

The callback is called with the error message. so this means that the callback is called with the error as first parameter. Whatever happens then is a topic of the node implementation that uses the library ... because exactyl the library is designed in a way to NOT throw any error.
If you want this error to be thrown then the node-red library need to do that

gitisgreat2023, 2025-05-06:
I use the Node Red package node-red-contrib-smartmeter

Can I somehow fix myself that I process the error message myself further, like catch does?

Apollon77, 2025-05-07:
I assume you would need to adjust it in https://github.com/coolchip/node-red-contrib-smartmeter/blob/master/smartmeter.js#L36 or reqquest a change by the developer of that project